diyhue/Lights #108: Custom diyHue Light Implementation

philib, 2021-01-19:
Hi,

i would like to synchronize my Philips TV with Ambilight+Hue feature with a RGB LED bar. For this I would like to use my Raspberry Pi. Accordingly, to realize the project, the Pi must act as both a hub and a light.

Unfortunately I have not found any specification of a diyHue light. Is there a SDK besides the existing firmware? If not, shouldn't it be enough to provide a web server on the local network with the '/detect' route (and the corresponding response) to be detected as a light by the hub? I have mocked the '/detect' route for now, but I could not find my "light" with the Hue Essentials app. Did I forget something?

Thanks in advance

philib, 2021-01-19:
Ok, this might be a problem i need to solve later on. Currently im running the light webserver mock on a different machine. Therefore there should not be port related issue. The emulator is able to find my machine in the network. I tested this with an nmap scan inside the container.

ryancasler, 2021-01-20:
Are you running DIYHue in docker in Bridge or Host network mode?  If Host, then you are going to run into problems.  If in Bridge, then use the bridge IP address to talk from your container to your host OS.  So, the IP address should be something like  172.17.0.1 within your host OS and then each container will get an IP in that subnet.  So, 172.17.0.2, etc.  That would be the best way if you are running in Bridge (default) networking mode.

philib, 2021-01-20:
@ryancasler does the network mode matter even if i'm currently mocking the light on a different **physical** machine in the local network? I tried the bridge mode, but also couldn't find my light mock either.

Im currently runnign a simple webserver serving json mocks on following routes
* http://192.168.0.5/detect
```
{
 name: "TestLight",
 protocol: "native_single",
 modelid: "LCT015",
 type: "rgb",
 mac: "74:70:fd:fb:54:32",
 version: 3.0,
}
```
* http://192.168.0.5/state
```
{ 
 on: true,
 bri: 144,
 xy: [0.53, 0, 21] 
}
```

Should'nt that be enough for the diyHue Emulator to find my "light"?

philib, 2021-01-23:
> The main issue is that port 80 is used by bridge emulator and cannot be used also by the light. You can manually add the light to config.json file of the bridge emulator. This way you can specify manually other free port.

@mariusmotea  I tried to change the port in the config.json manually but the emulator is not able to communicate with my light anymore. Am i missing something?

philib, 2021-01-29:
Hi again,
i hooked up a led lightstrip to my raspberry pi and am now able to control it via the Hue Essentials app. 
The next step was to try the entertainment mode. My first step was to create a mocked light agian listening to udp packets on the corresponding port 2100.  I successfully tested the "Sync" mode of the Essentials app using the camera sync feature (running the light server on my desktop pc).
You can find the code in my [repository](https://github.com/philib/diyHueNodeJsLight/blob/main/server.js).

However if i'm running the server on my raspberry pi and test the entertainment mode using camera sync, the camera preview 
of the entertainment app does not work properly. Also, the server does recieve some udp package now and then but rgb data remains always the same `{ r: 254, g: 255, b: 0 }`.

Do you have any idea what's causing this issue?

Thanks in advance

closed 2021-06-26 by stale[bot]