diyhue/Lights #109: No Update For WS2812 Sketch

ryancasler, 2021-03-08:
I replaced an old implementation with a new on so I thought I would get updated software as well.  What i am seeing is that if commands are issued directly on the device's web page, if you have more than one light defined the changes never show back up in DIYHue.  DIYHue will continue to show the last settings that it made but never poll for any updates to the light.  

Is there a way to push that information back to DIYHue from board?  A function/functions that can be called after the physical changes  are made?

mariusmotea, 2021-03-08:
There is an implementation currently to update the lights firmware from diyhue bridge (like original bulbs). This was the main reason why entire configuration was moved from arduino sketch to SPIFFS storage in json file (set by light web ui). It was tested with success but unfortunately automatic build of firmwares with Travis is not working. In next version of diyhue you will have many more options in the bridge web interface for better control.

ryancasler, 2021-03-08:
Yeah, that wasn't really what I was asking about.  I was talking about changes made tp the light  (like brightness) being made on the light itself not being reflected in DIYHue but i had to manually update my config file after updating the firmware.  I guess there were changes necessary that weren't being picked up by the old config.

closed 2021-03-08 by ryancasler

mariusmotea, 2021-03-11:
This already happens because the lights are fetched every 5 minutes for status and much faster if a hue app is opened and request lights status.