diyhue/Lights #110: Manual division of the lights for SK6812 strip

DarthAnwalt, 2021-04-10:
Hi!
First of all, many thanks for your work, project is absolutely awome.

I'm using SK6812 strip, and it looks like the dividible lights (pixels) function, which were added by @dobby5 to Generic_WS2812_Strip.ino sketch, is absent in Generic_SK6812_Strip.ino.
So, user can't manually choose number of leds for every virtual light with SK6812 light strip.

I tried to implement @dobby5 's code from Generic_WS2812_Strip.ino to Generic_SK6812_Strip.ino and tested result for successfully compiling and working with real hardware.
Could you please review this changes and, if all is right, update Generic_SK6812_Strip.ino in master branch.

Best regards!
[Generic_SK6812_Strip[1].txt](https://github.com/diyhue/Lights/files/6289620/Generic_SK6812_Strip.1.txt)

mariusmotea, 2021-04-10:
Hi,

Thanks for your work!. If you test this and is working good, please fork this repo, commit the changes to this file and create Pull Request to master, so your contribution will be recorded.

DarthAnwalt, 2021-04-11:
Ok, did it.
I'm new to Github, so I'm still learning how to use it.

Thanks for your reply!

closed 2021-04-11 by DarthAnwalt