diyhue/Lights #113: Generic Light Sketch has wrong default gateway address when changing to static

vKnmnn, 2021-05-06:
Hi Marius!

today i noticed a few shortcomings in the generic light sketches:

- I have a large amount of reconnects of my esp8266/esp8285 devices that get disconnected due to inactivity. 
Can you confirm this in your setup? If yes, can you implement some sort of keepalive at a coonfigurable interval so these devices will stay connected?

- I've wanted to set one of these devices to use a static IP address and noticed that in the Network tab, the Gateway field per default has an invalid address filled, namely the adress of the network itself, which cannot be the gateway.
[image]
It should be .1

- Upon correcting the gateway address and saving, it takes about 10 minutes for me to be able to communicate with the light again. Can you confirm whether this is an actual issue vs just my problems with wifi congestion or similar?

As always, thanks for your work on this!

mariusmotea, 2021-05-06:
Hi,

I use my home lights with both static and dynamic ip's. They join the netwok in about one second after reset/power on. They never disconnect even if i'm not home a few days. I think there is a network issue. I believe all fields are default from DHCP servers, check if your DHCP server has a bad configuration.

vKnmnn, 2021-05-06:
oh wow! how long has my router been running on .0 as gateway?!!? You were right, i fixed that now. My bad

closed 2021-05-06 by vKnmnn