diyhue/Lights #114: Unable to Boot on ADAfruit Feather HUZZAH ESP8266

Chicob13, 2021-06-02:
I was able to upload Tasmota to the ESP8266 and everything worked fine, and I wanted to create a lightstrip in HUE to reflect what was happening on my computer monitors, so I Tried to upload the Generic WS2812 Strip Sketch and this is what I get in the console when I reboot it:

`
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (0):
epc1=0x4020119c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3fff2280 end: 3fff2a60 offset: 0190
3fff2410:  40106730 feefeffe feefeffe 00200305  
3fff2420:  40004378 00000030 00000019 ffffffff  
3fff2430:  60000200 00000008 ffffffff 80000000  
3fff2440:  20000000 3fff3640 80000000 2020a060  
3fff2450:  80000000 00000005 00018cb3 3fff3644  
3fff2460:  000000a0 0020a060 00000100 00000030  
3fff2470:  00000000 400042db feefeffe feefeffe  
3fff2480:  40004b31 3fff34d0 00000100 003f8000  
3fff2490:  00000000 400042db feefeffe feefeffe  
3fff24a0:  00000002 4000444e 00000100 00200300  
3fff24b0:  00000000 400042db 60000200 3fff2524  
3fff24c0:  00000000 400042db 0000002c 00200400  
3fff24d0:  40004b31 3fff2510 00000004 0020042c  
3fff24e0:  40106730 00000001 3fff26a0 00200400  
3fff24f0:  40100206 00000001 3fff2510 00200400  
3fff2500:  3fffc718 0000002c 3fff26a0 4020e762  
3fff2510:  00000000 400042db 00000000 00200104  
3fff2520:  40004b31 3fff35e4 00000100 0020a000  
3fff2530:  40106730 00000001 3fff3098 00000001  
3fff2540:  40100206 00000001 3fff35e4 0020a000  
3fff2550:  00000001 00000100 00000000 4020e6bc  
3fff2560:  3fff27f8 4020e4d0 3fff35e4 4020e749  
3fff2570:  00200404 00000000 00000000 3fff25e2  
3fff2580:  40201204 3fff25e2 00200104 40213f6d  
3fff2590:  00000001 00000003 3fff2ea4 4000050c  
3fff25a0:  4000e240 00000030 0000001b ffffffff  
3fff25b0:  40215462 00000010 00000400 00000002  
3fff25c0:  00000012 00000000 00000000 40215434  
3fff25d0:  00000130 00000005 00018cb3 3ffeaa1e  
3fff25e0:  0000000b 00000100 3fff2ea4 00000030  
3fff25f0:  00200000 00000001 00000001 00200404  
3fff2600:  00200309 4020e4d0 00000000 3fff2ea4  
3fff2610:  0000002e 3fff3094 00000004 40216159  
3fff2620:  00000003 00000000 3fff2ea4 0000002e  
3fff2630:  0000002e 3fff26a0 00200400 40213f2d  
3fff2640:  3fff3394 00000001 3fff2ea4 402122c4  
3fff2650:  3fff2680 000000ff 000000ff 40217e3e  
3fff2660:  3fff3094 3fff329c 00008001 00200400  
3fff2670:  00000100 0000001d 3fff2ea4 4021229c  
3fff2680:  00010004 000000fc 3fff2ea4 00000100  
3fff2690:  00000100 3fff35e4 0020a000 40213f2d  
3fff26a0:  3fff3394 00000000 3fff2ea4 402122c4  
3fff26b0:  6769666e 6f736a2e 0000006e 40217eb4  
3fff26c0:  3fff35e4 00000100 3fff35d0 0020a000  
3fff26d0:  3fff3194 00000004 00040003 80010001  
3fff26e0:  3fff3094 0000011f 00000020 0000a000  
3fff26f0:  00000100 3fff2ea4 00000005 402157b5  
3fff2700:  00000000 00000000 00000000 00000080  
3fff2710:  3fff31d2 00000000 00001e07 00000000  
3fff2720:  00001e06 00000000 40215434 3fff3194  
3fff2730:  00000000 00000100 3fff2755 40207040  
3fff2740:  3fff329c 00000020 3fff37f8 3fff27b6  
3fff2750:  3ffeaa1e 00000008 3fff2ea4 40217182  
3fff2760:  3ffeaa1e 00000000 3fff2774 3fff2770  
3fff2770:  0000001e 00000004 3fff3d1c 40100a6e  
3fff2780:  3fff27b0 3ffe85e8 3fff3d1c 402274a8  
3fff2790:  000001d5 000001d5 3ffe8624 3fff2ea4  
3fff27a0:  3ffeaa1e 00000008 3ffeaa1e 402144f5  
3fff27b0:  3fff32cc 00000020 3fff3914 40100a6e  
3fff27c0:  00000000 3ffe85e8 00000001 40203128  
3fff27d0:  3fffdad0 3ffe85e8 00000001 3fff2ea4  
3fff27e0:  3fff2e9c 3fff285c 3ffeaa1e 402129e1  
3fff27f0:  00000000 3fff3d28 3fff3d1c 40212750  
3fff2800:  00000000 00000008 00000000 3fff1224  
3fff2810:  3fff3914 3fff3952 3fff3b24 3fff1224  
3fff2820:  3ffeaa1e 3fff12a8 3fff2880 4020e92c  
3fff2830:  3fff3b64 3fff2810 00000000 3fff3bb4  
3fff2840:  3fff2810 00000000 3fff3c04 3fff2810  
3fff2850:  00000000 00000001 3fff2870 40202899  
3fff2860:  3fffdad0 00000001 00000000 40208307  
3fff2870:  3fffdad0 3ffe85e8 3fff10e4 402084b8  
3fff2880:  00000000 00000000 00000000 00000000  
3fff2890:  00000000 00000000 3fff12a8 00000578  
3fff28a0:  000000f0 3fff2ea4 3fff2e9c 40212c45  
3fff28b0:  00000578 4021c39c 3fff2ea4 402159bc  
3fff28c0:  00000007 00000007 00000001 40203408  
3fff28d0:  60000b24 00001000 60000b28 40203191  
3fff28e0:  00000000 3fff3914 00000020 3fff3918  
3fff28f0:  00000000 3fff3914 000002d0 402035bd  
3fff2900:  3f000000 3fff2ea4 3fff2e9c 40212cf4  
3fff2910:  40211c61 00000000 3fff12a8 3fff1224  
3fff2920:  3fffdad0 3ffe85e8 3fff10e4 402098fb  
3fff2930:  feefeffe feefeffe feefeffe feefeffe  
3fff2940:  feefeffe feefeffe feefeffe feefeffe  
3fff2950:  feefeffe feefeffe feefeffe feefeffe  
3fff2960:  feefeffe feefeffe feefeffe feefeffe  
3fff2970:  feefeffe feefeffe feefeffe feefeffe  
3fff2980:  feefeffe feefeffe feefeffe feefeffe  
3fff2990:  feefeffe feefeffe feefeffe feefeffe  
3fff29a0:  feefeffe feefeffe feefeffe feefeffe  
3fff29b0:  feefeffe feefeffe feefeffe feefeffe  
3fff29c0:  feefeffe feefeffe feefeffe feefeffe  
3fff29d0:  feefeffe feefeffe feefeffe feefeffe  
3fff29e0:  feefeffe feefeffe feefeffe feefeffe  
3fff29f0:  feefeffe feefeffe feefeffe feefeffe  
3fff2a00:  feefeffe feefeffe feefeffe feefeffe  
3fff2a10:  feefeffe feefeffe feefeffe feefeffe  
3fff2a20:  feefeffe feefeffe feefeffe feefeffe  
3fff2a30:  feefeffe feefeffe feefeffe 3fff1224  
3fff2a40:  3fffdad0 00000000 3fff1210 4021166c  
3fff2a50:  feefeffe feefeffe 3ffe8620 40100d21  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v0006af20
~ld

`

I don't know if there is a library that I have that is more up to date, but it appears that something on my side is not uploading right

mariusmotea, 2021-06-03:
Avoid the new esp8266 library v3.0.0. use latest V2 library instead.

Chicob13, 2021-06-03:
And that Fixed the issue, thank you, I figured there was a library i was using that was either too new or out of date but i couldnt pin point it, Thank you again!

closed 2021-06-03 by Chicob13