diyhue/Lights #121: SK6812 Firmware keeps rebooting/disconnecting.

Justdigit, 2021-11-07:
I have tried to flash the SK6812 firmware but i cant get it to work.
The Wifi acces point goes well. When the esp optained the ip adress i can get in the webgui.
But its for really short period. It reboots itself but the second try after rebooting it wont load the webpage.

I have tried several esp8266 ...... did blank bin flash. When i flash the WS2812 firmware al works well.
I

Justdigit, 2021-11-07:
Turns out there was a very strange issue with my router....
Still strange WS2812 firmware was no problems.

closed 2021-11-07 by Justdigit

mariusmotea, 2021-11-07:
Most common problems are caused by lack of SPIFFS partition or flash space. If you use Arduino ensure there is set some kb space dedicated for SPIFFS before you flash the controller.