diyhue/Lights #122: My diyhue-ESP light is not found by my original Philips Hue Bridge

TimW2022, 2022-01-03:
Today I managed to build the app from

https://github.com/diyhue/Lights/tree/master/Arduino/Generic_RGB_CCT_Light

and got it flashed on ESP8266-ESP12F module. All 5 PWM outputs are operational and controllable via the ESP websever GUI. That is great. But my Philips Hue-Bridge is not detecting any new lamp. The new ESP-diy-lamp has stable wifi connection and is controllable via Web, also during the search.

Is there anything I need to add or to configure in the app that it can be found by the Philips-Hue-Bridge ?

The philiphs hue bridge detects my (non philips) zigbee lamps fine, but I never tried this with a wifi lamp. I assume it polls out over Ethernet and my ftirtzbox-Wifi-AP to search wifi lamps in parallel to zigbee lamps, but I don't know. Will go for some wireshark work the next days.

Thanks in advance
Tim

Mevel, 2022-01-03:
Diyhue lights will never work with original hue bridge.
Not just because the vendor doesn't like it but simply because it uses two different communication technologies.

That's what the diyhue emulator is for. It emulates a original hue bridge with the benefits of using a variety of devices across vendors.

You can start your own instance and add the diyhue light there.

Enjoy

TimW2022, 2022-01-04:
Thanks for the fast feedback Mevel !

closed 2022-01-04 by TimW2022