diyhue/Lights #124: [beta] Reduce or disable logging to syslog

SvenGWK, 2022-02-16:
Hello,

I have installed the diyhue beta as integration in Home Assistant (https://github.com/MaxBec/diyHue-hassio).
Unfortunately I can't find a way to stop diyhue from logging to the syslog file.
The file is flooded with such messages:

```
Feb 16 20:51:35 central 3f2eee367733[813]: A
Feb 16 20:51:35 central 3f2eee367733[813]: {'alert': 'select',
Feb 16 20:51:35 central 3f2eee367733[813]:  'bri': 166,
Feb 16 20:51:35 central 3f2eee367733[813]:  'colormode': 'xy',
Feb 16 20:51:35 central 3f2eee367733[813]:  'ct': 500,
Feb 16 20:51:35 central 3f2eee367733[813]:  'mode': 'homeautomation',
Feb 16 20:51:35 central 3f2eee367733[813]:  'on': False,
Feb 16 20:51:35 central 3f2eee367733[813]:  'reachable': False,
Feb 16 20:51:35 central 3f2eee367733[813]:  'xy': [0.599, 0.382]}
Feb 16 20:51:35 central 3f2eee367733[813]: A
Feb 16 20:51:35 central 3f2eee367733[813]: {'alert': 'select',
Feb 16 20:51:35 central 3f2eee367733[813]:  'bri': 173,
Feb 16 20:51:35 central 3f2eee367733[813]:  'colormode': 'xy',
Feb 16 20:51:35 central 3f2eee367733[813]:  'ct': 500,
Feb 16 20:51:35 central 3f2eee367733[813]:  'mode': 'homeautomation',
Feb 16 20:51:35 central 3f2eee367733[813]:  'on': False,
Feb 16 20:51:35 central 3f2eee367733[813]:  'reachable': False,
Feb 16 20:51:35 central 3f2eee367733[813]:  'xy': [0.599, 0.382]}
Feb 16 20:51:35 central 3f2eee367733[813]: A
Feb 16 20:51:35 central 3f2eee367733[813]: {'alert': 'select',
Feb 16 20:51:35 central 3f2eee367733[813]:  'bri': 166,
Feb 16 20:51:35 central 3f2eee367733[813]:  'colormode': 'xy',
Feb 16 20:51:35 central 3f2eee367733[813]:  'ct': 500,
Feb 16 20:51:35 central 3f2eee367733[813]:  'mode': 'homeautomation',
Feb 16 20:51:35 central 3f2eee367733[813]:  'on': False,
Feb 16 20:51:35 central 3f2eee367733[813]:  'reachable': False,
Feb 16 20:51:35 central 3f2eee367733[813]:  'xy': [0.599, 0.382]}
```

I have tried to change in [logger.py](https://github.com/diyhue/diyHue/blob/beta/BridgeEmulator/logManager/logger.py) loglevel = logging.DEBUG to logging. WARNING and then rebuilt everything, but unfortunately this did not lead to any result.
Is there anything else I could do?

mariusmotea, 2022-02-16:
Can you somehow set ENV variable `DEBUG=false` ? I don't use Home Assistant plugin, but i know the addon is a docker container and we implemented DEBUG env flag to enable debug mode.

SvenGWK, 2022-02-16:
Thanks for the tip, I'll take a look at how that could work.
And sorry that I posted the question under /lights. I had unfortunately not noticed that I'm not in the main repo.

closed 2022-04-18 by stale[bot]