diyhue/Lights #125: Arduino nano 33 IoT support

Obi-Stan-Kenobi, 2022-02-17:
I was wondering if it is possible to make the Gradien Strip code work on an Arduino nano 33 IoT.

mariusmotea, 2022-02-17:
Unfortunately we don't maintain the entire code, we rely on certain libraries so if these libraries are supporting also Arduino nano 33 IoT it must work. Did you tried to compile the code and get errors? I think a small change is require to NeopixelBus library to specify the strip data in pin (on ESP8266 is not specified since just RX pin is working) and also change wifi connection code with the one for Arduino nano 33 IoT (probably you need to remove WiFI Manager library that is made for ESP8266). Not having such board is hard for me to change the code for it.

mariusmotea, 2022-02-17:
Look in the other sketches where SPIFFS is still used. ESP library migrated to LittleFS and consider SPIFFS deprecated, for this reason we did the switch. In the sketch you need to rename LittleFS with SPIFFS and in the header import FS.h instead of LittleFS.h

mariusmotea, 2022-02-17:
Look, this is still using SPIFFS: https://github.com/diyhue/Lights/blob/master/Arduino/Generic_WS2812_Strip/Generic_WS2812_Strip.ino

closed 2022-04-18 by stale[bot]