diyhue/Lights #126: sk6812 gradient

DeLoWaN, 2022-03-17:
Hi !

As we already have the support of gradient light to ws2812, it would be interesting to have it also on sk6812.

I wanted to also use these strip as main lightning on my room, hence the choice of sk6812 which provides a white LED.

What can be the effort of creating a sketch for this ? Thanks.

mariusmotea, 2022-03-17:
Hi, the issue is that i currently without any SK6812 strip to test. Are you ok to test such sketch? If yes, can you join slack channel to have interactive talk?

DeLoWaN, 2022-03-17:
Yes sure ! Right now I don't have the hardware since I was reviewing all the solutions that exists to acheive what I want. Gradient would be a great addition though not a must have in my setup, so I will buy the hardware and I let you know when I will have it available for testing.

mariusmotea, 2022-03-17:
I made an initial release. Please provide me your feedback once you have the hardware. https://github.com/diyhue/Lights/tree/master/ESP8266/Hue_Gradient_Lightstrip_SK6812

Mevel, 2022-03-23:
@DeLoWaN did you have the chance to test the firmware? 
can you give us some feedback?

DeLoWaN, 2022-03-23:
@Mevel as discussed with @mariusmotea on [Slack](https://diyhue.slack.com/archives/CC74V7HJL/p1647885324239359), this is working so far.
However, from the Sketch @mariusmotea posted, it still need `NeoSk6812Method` to be replaced by `Neo800KbpsMethod` for this to work properly.

Did not have the chance yet to test it with the Hue App.

closed 2022-04-18 by stale[bot]