diyhue/Lights #128: Color change

MCZERNIK10, 2022-05-24:
Hi,
In file:
ESP8266)/Generic_WS2812_Strip/

I have a problem, I cannot redefine the colors in lines 53 and 613 of NeoGrbFeature to NeoBgrFeature.
Arduino generates compile errors.

mariusmotea, 2022-05-25:
Hi,

can you paste here the error lines? Try to downgrade WiFi Manager library, there are known issues with recent versions.

MCZERNIK10, 2022-05-25:
Hi.
Generic_WS2812_Strip:53:13: error: 'NeobgrFeature' was not declared in this scope

 NeoPixelBus<NeobgrFeature, Neo800KbpsMethod>* strip = NULL;

             ^

Generic_WS2812_Strip:53:44: error: template argument 1 is invalid

 NeoPixelBus<NeobgrFeature, Neo800KbpsMethod>* strip = NULL;

                                            ^

Generic_WS2812_Strip:53:53: error: invalid type in declaration before '=' token

 NeoPixelBus<NeobgrFeature, Neo800KbpsMethod>* strip = NULL;

                                                     ^

C:\Users\Czerniki\Desktop\Lights-master\ESP8266\Generic_WS2812_Strip\Generic_WS2812_Strip.ino: In function 'void infoLight(RgbColor)':

Generic_WS2812_Strip:219:12: error: request for member 'SetPixelColor' in '* strip', which is of non-class type 'int'

     strip->SetPixelColor(i, color);

            ^

Generic_WS2812_Strip:220:12: error: request for member 'Show' in '* strip', which is of non-class type 'int'

     strip->Show();

            ^

Generic_WS2812_Strip:222:12: error: request for member 'SetPixelColor' in '* strip', which is of non-class type 'int'

     strip->SetPixelColor(i, black);

            ^

Generic_WS2812_Strip:223:12: error: request for member 'Show' in '* strip', which is of non-class type 'int'

     strip->Show();

            ^

C:\Users\Czerniki\Desktop\Lights-master\ESP8266\Generic_WS2812_Strip\Generic_WS2812_Strip.ino: In function 'void lightEngine()':

Generic_WS2812_Strip:318:24: error: request for member 'SetPixelColor' in '* strip', which is of non-class type 'int'

                 strip->SetPixelColor(pixel, convFloat(lights[light].currentColors));

                        ^

Generic_WS2812_Strip:320:24: error: request for member 'SetPixelColor' in '* strip', which is of non-class type 'int'

                 strip->SetPixelColor(pixel, blending(lights[0].currentColors, lights[1].currentColors, pixel + 1 - (dividedLightsArray[0] - transitionLeds / 2 ))); // calculate the transition led color 

                        ^

Generic_WS2812_Strip:339:24: error: request for member 'SetPixelColor' in '* strip', which is of non-class type 'int'

                 strip->SetPixelColor(pixel + pixelSum + transitionLeds, blending( lights[light - 1].currentColors, lights[light].currentColors, pixel + 1));

                        ^

Generic_WS2812_Strip:343:24: error: request for member 'SetPixelColor' in '* strip', which is of non-class type 'int'

                 strip->SetPixelColor(pixel + pixelSum + transitionLeds, blending( lights[light].currentColors, lights[light + 1].currentColors, pixel + transitionLeds / 2 - dividedLightsArray[light]));

                        ^
[... 97 lines cut ...]
               strip->SetPixelColor(pixel + pixelSum + transitionLeds, blendingEntert( lights[light - 1].currentColors, lights[light].currentColors, pixel + 1));

                      ^

Generic_WS2812_Strip:952:22: error: request for member 'SetPixelColor' in '* strip', which is of non-class type 'int'

               strip->SetPixelColor(pixel + pixelSum + transitionLeds, blendingEntert( lights[light].currentColors, lights[light + 1].currentColors, pixel + transitionLeds / 2 - dividedLightsArray[light]));

                      ^

Generic_WS2812_Strip:955:22: error: request for member 'SetPixelColor' in '* strip', which is of non-class type 'int'

               strip->SetPixelColor(pixel + pixelSum + transitionLeds, convFloat(lights[light].currentColors));

                      ^

Generic_WS2812_Strip:961:16: error: request for member 'ClearTo' in '* strip', which is of non-class type 'int'

         strip->ClearTo(convFloat(lights[light].currentColors), 0, pixelCount - 1);

                ^

Generic_WS2812_Strip:964:12: error: request for member 'Show' in '* strip', which is of non-class type 'int'

     strip->Show();

MCZERNIK10, 2022-05-25:
When compiling with NeoBrgFuture, the compiler reports no errors but the green and red colors are inverted on the palette.

mariusmotea, 2022-05-26:
Can you show the libraries versions you are using. I manage to compile the code with all new libraries. Do you have neopixelbus by Makuna library installed?

caribo, 2022-05-26:
Check your code .

Shouldn't it be NeoBgrFeature rather than NeobgrFeature?

mariusmotea, 2022-10-01:
This was a naming issue in WiFI Manager library that was fixed in recent releases.

closed 2022-10-01 by mariusmotea