diyhue/Lights #135: Unable to download and flash via Web

rowbotik, 2022-10-30:
I have an ESP8266 with WS2812B. I had flashed it with WLED and it all ran fan so I decided to switch to DIY Hue. I attempted multiple times to install any of the Lightstrip options but they all give an error downloading. I ended up downloading the .bin file and installed with ESP flasher but wasn't able to use the diyhue portal. I tried 2 different machines and same issue.

Mevel, 2022-10-30:
I will try to replicate the error tomorrow.
Any specific logs or error messages?

What esp8266 variant do you use?

rowbotik, 2022-10-30:
`Downlading firmware https://raw.githubusercontent.com/diyhue/Lights/master/Arduino/bin/undefined failed: 404`

Using a ESP-12E? I manage to get them flashed the .bin file but never manages to get the other LEDs on when editing the pixel count. Is there some special background work that happens when using the .bin files with the online tool?

[image]
[image]

danflow, 2022-11-18:
Hi, I have the same problem with the equal dev board. Thanks for a possible resolving.

kieranjones3939, 2022-12-06:
I am having similar issues, only the TV Gradient firmware seems to be working, when flashing with ESP flasher this is the only firmware that produces logs after flashing.

Papa-T6, 2022-12-09:
I'm also getting the download error. I'm using an ESP8266 and trying to select the ON_OFF v2 light type. Any updates would be greatly appreciated!

easaw, 2023-08-20:
[make sure you configure the CORS-headers](https://esphome.github.io/esp-web-tools/#:~:text=make%20sure%20you%20configure%20the%20CORS%2Dheaders)