diyhue/Lights #136: Issues changing individual lights on WS2812 gradient strip via lights API

Arlodotexe, 2022-11-26:
## Background

I have DIYHue set up with [this](https://www.amazon.com/dp/B01CDTECSG) WS2812b strip and an ESP8266. 
Everything seems to be working normally. The emulated bridge is slow but it works, and can set scenes with gradients.

Output from the `detect` endpoint:
```json
{"name":"Gradient Strip 1","protocol":"native_single","modelid":"LCX002","type":"ws2812_gradient_strip","mac":"redacted","version":4}
```

## Problem

Changing the entire strip works as expected:
```bash
curl 'http://192.168.199.51/state' -X 'PUT' --data '{"on":true,"bri":144,"xy":[0.53, 0,21]}';
```

But changing multiple/individual lights does not: 
```bash
 curl 'http://192.168.199.51/state' -X 'PUT' --data '{1: {"on":true,"bri":144,"xy":[0.53, 0,21]}, 2: {"on":true,"bri":144,"ct":370}, 3: {"on":false}}';
```

mariusmotea, 2022-11-27:
Hi.

The json body must be based on Hue api and you can check how it is parsed here: https://github.com/diyhue/Lights/blob/master/ESP8266/Hue_Gradient_Lightstrip_WS2812/Hue_Gradient_Lightstrip_WS2812.ino#L692-L701

So it must look like: `{"gradient": {"points": [ {"color": {"xy": {"x": 0.53, "y":  0.21}}}, {"color": {"xy": {"x": 0.13, "y":  0.2}}}, {"color": {"xy": {"x": 0.3, "y":  0.3}}}]}}`

Note that the number of segments is dynamically updated based on the number of "points" from the request witch is a list, in my example are 3, but can be between 2 up to 10.

P.S. The emulated bridge must not be slow, here probably you have another issue.

Arlodotexe, 2022-11-27:
I was poking through this source code quite a bit and saw this was possible, although I tried it and it caused a failure that rebooted the device for some reason.

The provided JSON does work though. Must have been a syntax error. Thank you @mariusmotea! Maybe worth adding to the docs?

Arlodotexe, 2022-11-27:
Wait, gradients work, but I still need the ability to change individual LEDs for custom effects. It doesn't look like this is possible via the API, would I need to modify and compile my own firmware?

closed 2022-11-27 by Arlodotexe

Mevel, 2022-11-27:
LEDs are grouped into virtual lights. There is no per pixel manipulation via api.

mariusmotea, 2022-11-27:
You can implement custom effects intro the firmware, check fireplace and candle effects that are already there. If you manage to add nice effects feel free to open a pull request with your changes.

Arlodotexe, 2022-11-28:
It may be more appropriate to say that I'm trying to create _dynamic_ effects using external sensors. Only illuminating where a person is standing, for example, without losing Hue compatibility.