diyhue/Lights #137: Web flash tool unable to download firmware

Papa-T6, 2022-12-09:
I'm getting a firmware download error using the web flash tool. I'm using an ESP8266 and trying to select the ON_OFF v2 light type. Any updates would be greatly appreciated!

Is this just a matter of a broken URL? The web flash tool is trying to download from...
https://raw.githubusercontent.com/diyhue/Lights/master/Arduino/bin/Generic_ON_OFF_Light.bin
That url looks dead.

_Originally posted by @Papa-T6 in https://github.com/diyhue/Lights/issues/135#issuecomment-1343751671_

Mevel, 2022-12-09:
Checking on it

easaw, 2023-08-20:
@Mevel this is still an issue.

Reason is CORS https://esphome.github.io/esp-web-tools/#:~:text=configure%20the%20CORS%2Dheaders

[If you prefer to locally host the JavaScript](https://esphome.github.io/esp-web-tools/#:~:text=If%20you%20prefer%20to%20locally%20host%20the%20JavaScript)