diyhue/Lights #139: Feature Request: New Sketch: Gradient & Gradient TV RGBCCT

TRusselo, 2023-02-12:
There is a RGBBCCT (RGBWW) strip sketch for a solid color strip
But the Gradient & Gradient TV sketches are based on RGB  ( no white/warmwhite)
would be great if we could get RGBCCT gradient sketches.  (TV and non-TV gradient sketches)

the white color matching between the DIY and official are way off due to official hue using RGBCCT 6-pin strips.

mariusmotea, 2023-02-12:
Gradient effect can be created only on the strips with addressable leds. I think you want a sketch for SK6812 leds that have a dedicated powerful white led. Currently we have the regular Gradient Lightstrip sketch for SK6812, but not the TV Gradient .

Paalap, 2023-02-12:
There are also FW1906 stips in place which are addressable and RBGCCT but the usual color conversion would not use RGB and CCT at the same time. 

Was thinking about to try these (and to adjust the sketches accordingly)

mariusmotea, 2023-02-12:
I look to see if there is support for FW1906 in the NeopixelBus library that we are using, and it seams it did not support these leds. I will take a look on these since i like they are using 24V power and may be suitable for a project of mine, but this will require some time.

https://github.com/Makuna/NeoPixelBus

Paalap, 2023-02-12:
It is the same approach instead of 3 or 4 bits there are 5 (actually 6, RGB + empty + cw + ww) sent. Thus neopixel would require just a small adjustment but I would like to use also CCT on RGB mix and not only in CT mode - still searching for this

mariusmotea, 2023-02-13:
It require small improvement on this PR #129 . There was added support for rgb to rgbw conversion. I test it with SK6812 and i work.

Paalap, 2023-05-20:
@mariusmotea did you started some work here?

There is some initial development at NeopixelBus https://github.com/Makuna/NeoPixelBus/issues/702

But just the initial draft and is not working for me...

iceteaSA, 2025-07-02:
@mariusmotea The WS2805 LEDs are another example of RGB CCT strips with ICs. Would be awesome if support is added for them.

mariusmotea, 2025-07-03:
I need to have the WS2805 first, but i don't expect to be complicated to merge the WS2812 sketch with the RGB-CCT sketch.

iceteaSA, 2025-07-03:
I am going to get some soonish, for other projects, I would be happy to test? I want to use the BTL SP530E controller, but I would probably need a ESP32 sketch? I am going to install WLED for other use cases, as it already supports them. I doubt it is possible to use WLED lights as gradients?