diyhue/Lights #141: WS2812 + MOSFET

physicsG, 2023-06-14:
The sketches for WS2812 contain a reference to using a MOSFET to cut power to the LED strip. I cannot find a single piece of documentation on how to wire such MOSFET in order to get it to work or which MOSFET I can use in general. Can someone please elaborate on this?

This is the line of code:
`#define POWER_MOSFET_PIN 13 // WS2812 consume ~1mA/led when off. By installing a MOSFET it will cut the power to the leds when lights ore off.`
Later in code:
`  cutPower(); // if all lights are off GPIO12 can cut the power to the strip using a powerful P-Channel MOSFET`

mariusmotea, 2023-06-14:
Hi, Check the two transistors in top-right corner:

https://github.com/diyhue/Lights/blob/master/ESP32/ESP-C3_Controller_Board/schematic.pdf

The POWER_MOSFET_PIN is any free GPIO pin you can connect to toogle the mosfet.

physicsG, 2023-06-14:
Thanks for pointing to some documentation :) I feel that for my case it should be simpler as I'm using a Wemos D1 Mini (ESP8266). Do you know of a simpler diagram/use-case?

mariusmotea, 2023-06-15:
You need only Q1,Q2 and R3 from that schematic. You can also try using just one transistor (the power MOSFET) but you will cut the ground in this case (not the +5V power line) and this may result in some 5V returning on the DATA IN pin of the strip directly to ESP8266

ryancasler, 2023-06-16:
You could also use a relay instead on the 5v line.   This would cause an audible click as a relay is a physical switch.  Or you could use a P-channel Mosfet on the 5v line as well.

Fisico, 2023-06-17:
Keep in mind if you don't use smd mosfets you want to select one with as little internal resistance as possible. Otherwise you need a cooler for it.

physicsG, 2023-06-19:
Thanks for thinking with me here. Sorry for hijacking an issue tracker for this.

I happen to have an IRF640N, which is an N-channel mosfet. I tried to wire it as follows:

Gate: Connected to digital pin GPIO13 (D7) on the Wemos D1 mini.
Drain: Connected to the negative terminal of my WS2812 LED strip.
Source: Connected to the ground (negative) of my power supply.

I'm using the generic WS2812 strip ESP8266 code here: https://github.com/diyhue/Lights/tree/master/ESP8266/Generic_WS2812_Strip

Wiring it normally without transistor works fine and gives me the normal led control. When using the scheme above with the transistor, the first led in my led strip of 18 leds turned on very dimly, but there is no control whatsoever. 

What could I be doing wrong? Am I missing a resistor somewhere? The code uses POWER_MOSFET_PIN 13.

mariusmotea, 2023-06-20:
First you should measure with a multimeter if D7 pin voltage change from 3.3V to 0V when you toggle the light. This will indicate that everything is ok at ESP level and the firmware and the issue is with the mosfet wiring.

ryancasler, 2023-06-23:
Also, you should be using a resistor to pull the gate to ground when it is not being pushed high by the D1 mini.    You can look at this one with regards to how the Mosfet should be wired:
https://github.com/diyhue/Lights/tree/master/ESP8266/Generic_Dimmable_Light