diyhue/Lights #143: Hue Gradient led strip type

Ceda, 2023-08-05:
I am looking for the answer to the question, what type of LED strip is used for the HUE LED White and Color Ambiance Gradient. The original strip broke and I have the controller left, I would like to replace the strip and use it with the original controller.
Is there any compatible LED strip, or does Philips use its own chips for these strips?

Thank you for any information.

mariusmotea, 2023-08-05:
I don't see very well the marking of the IC. It is D55E DBGESA ?

mariusmotea, 2023-08-05:
May be this: https://www.advateklights.com/knowledge-base/ucs2904

Ceda, 2023-08-05:
> I don't see very well the marking of the IC. It is D55E DBGESA ?

Is 05EE DBGESA

Ceda, 2023-08-05:
> May be this: https://www.advateklights.com/knowledge-base/ucs2904

Thank you, I am exploring it ;)

Ceda, 2023-08-05:
It seems that UCS2904 can only control RGBW, but the IC on the hue strip controls RGBCCT.

mariusmotea, 2023-08-05:
This is wired, the IC in your picture has 8 pins. RGB + CCT require 5 pins (3 RGB + 2 CCT) so for data and power remain just 3. For power must be 2 so it remain just one for data in/out. Is possible that this IC control just the RGB led and CCT leds are controlled by some mosfets inside the control board since here there is no gradient. If this is true the IC may be also WS2811.

Ceda, 2023-08-05:
[image]
I tried to measure the connection and all the diodes lead to the IC (RGB WW CW), 4 inputs are used to connect the led strip. GND/VCC/DATA and 4ty? I have no clue. I will try to unsolder the leds and resistors and measure even better.

BasF0, 2023-12-29:
I don't think that there is any third party led strip on the market that works the same. Anyway, here is some more technical info:

There are only 3 electrical connections, GND, VCC, DATA. The fourth connection is not connected, as the marking on the PCB indicates. Therefore, all 5 RGB+CW+WW channels must be driven by the chip. It seems that the chip is not fed off of the 24V directly, the highest voltage of the chip's pins is about 21.2V (while there is 24V on the strip). 

As far as I can see (I used a very basic oscilloscope) data is always sent at 5*16+8 bits. Logic 1 is encoded as a PWM signal of higher pulse width, logic 0 a low pulse width (scope is not fast enough to properly measure width). 1 transmission every 10ms. The order of the data is as follows:
red, green, cold-white, blue, warm-white, +8 low duty cycle pulses
Logic level of data is 5V. The 8 pulses are unknown to me, their duty cycle stays the same when the brightness is adjusted. 

Regarding the pinout, I think it is as follows, going in a circle around the chip (not a 100% sure on them though)
1 (where the chip marker is): data
2: must be ground ? see text below, pulses on every data transmission
3: cold white
4: green
5: red
6: blue
7: warm white
8: VDD (not 24V ?)
Interestingly; I see that pin 2 of the chip pulses exactly and only on the last 8 pulses of each transmission (5V). (Note: my measurements were done on a strip containing only 2 chips, on the second chip).

Don't know (yet) how the different segments of the strip are identified when using it in actual gradient mode (where different sections of the strip are colored differently).