diyhue/Lights #144: Self-compiled sketches == reset loop

lukeIam, 2023-09-17:
Hi,

I have a SK6812 led stripe and an esp8266 (D1 Mini) and I want to use it in hue entertainment.
So I'm trying to compile this: (as there seems to be no binaries for  SK6812_gradient):
https://github.com/diyhue/Lights/blob/master/ESP8266/Hue_Gradient_Lightstrip_SK6812/Hue_Gradient_Lightstrip_SK6812.ino

Compiling is not the problem (succeeds without errors) but when I flashed it to the esp8266 I get into a reset loop.
I also tried the same with an ESP32 -> same problem (with the esp32 sketch).

If I flash over the website it works fine (but no gradient available).

Any idea?

mariusmotea, 2023-09-18:
Hi,

What libraries do you use? Probably one is wrong.

lukeIam, 2023-09-18:
> What libraries do you use? Probably one is wrong.

Most likely - I played around with different versions but no success (is there an official list with the correct versions?)
I'm currently using:
[image]

Compiling works fine...

mariusmotea, 2023-09-19:
Have you flashed with SPIFFS partition to allow creation of config file?

lukeIam, 2023-09-19:
Yes - followed the instructions I found somewhere here in the repo.

BTW: is there a technical limitation that your web flasher only supports some sketches?

lukeIam, 2023-09-21:
this is what I get via serial:
```
��n�r��n|�l�l`bbrl�nb�nl`�rl�l��
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v0006a580
~ld
```

lukeIam, 2023-09-21:
It looks like with the flash size option: `Mapping defined by Hardware and Sketch` it's working ^^

closed 2023-09-21 by lukeIam