diyhue/Lights #146: Arduino IDE / Flashing config help needed, just getting resets right now.

csprocket777, 2024-01-12:
Hi, This is my first time doing anything with these sorts of components and need some help ensuring I have the right flashing configurations.

## Goal:
To make a Philips Hue compatible gradient light strip for use with my TV.

## My hardware components:
- [ESP8266 NodeMCU (ESP-12E)](https://www.amazon.com/gp/product/B010O1G1ES/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1)
- [ESP32-C3-DevKitC-02](https://www.amazon.com/gp/product/B09D3S4RPZ/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&th=1)
- [BTF-Lighting SK6812 16' light strip](https://www.amazon.com/gp/product/B01N2PC9KK/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1)
- [5V 20A 100W Power Supply](https://www.amazon.com/Aclorol-Switching-Universal-Transformer-Converter/dp/B07KC55TJF?crid=17B69B8MP05SH&keywords=5v%2Bpower%2Bsupply%2B20a&qid=1680301889&sprefix=5v%2Bpower%2Bsupply%2B20a%2Caps%2C84&sr=8-4&linkCode=sl1&tag=zacbuilds07-20&linkId=fa8d1da4c66c63c619d785bd3b681dc5&language=en_US&ref_=as_li_ss_tl&th=1)

## For the ESP8266
I'm using the [Hue_Gradient_Lightstrip_SK6812.ino](https://github.com/diyhue/Lights/tree/master/ESP8266/Hue_Gradient_Lightstrip_SK6812) file from the repo.

**My Arduino IDE settings:**
[image]

**Result:**  
It boots up just fine, I can connect it to my Wifi, I can even control it using the WebUI. But it defaults to a pixel count of 60 which means only a small portion of the strip is actually used. 

**Problem:**
When I use the WebUI to configure it to use the full 300 pixels in the strip, it gets into a reboot loop and never stabilizes.

## For the ESP32-C3:
I'm using the [ESP32_C3_SK6812_Gradient_Strop.ino](https://github.com/diyhue/Lights/tree/master/ESP32/ESP32_C3_SK6812_Gradient_Strip) file from the repo.

**My Arduino IDE settings:**
[image]

**Result:**
Similar problem to the ESP8266, it boots up just fine, I can connect it to my Wifi, I can even control it using the WebUI. But it defaults to a pixel count of 60 which means only a small portion of the strip is actually used. 

**Problem:**
Again, a similar problem, when I use the WebUI to configure it to use the full 300 pixels in the strip, it reboots and then appears as another name in the network (something like "esp32c3_something") and I can't get back to the WebUI.

In both cases, I see the initial white pulse in the strip and then the green pulse. The ESP8266 repeats those pulses on every reboot, the ESP32-C3 does not, it only happens once.

**My suspicion:**
Either:
- That I have my flashing settings incorrect
- I'm suffering a voltage drop

In either case, I'm not sure of the solution. Any help would be really appreciated.

mariusmotea, 2024-01-13:
Hi,

When you apply a configuration change in the web ui, the software generate a json file with that config, write it to FS storage and reset. After reset it will load the json file and run with it. The issue you may have is lack of FS storage (but from your images i see 3MB FS) or invalid/corrupt json file. Is the same issue happening even if you don't have the leds connected? If you have small PS when enable more leds (than 60) it may not be able to sustain the current load.

csprocket777, 2024-01-13:
Thank you for responding!

From what I've been reading, this strip takes ~0.06amps per LED, 300 leds = ~18 amps. I'm using a 20 amp power supply at 5V. Which is what confuses me.

Right now, on the ESP32-C3, I have nothing else connected, just power and USB, I just flashed it again, I'm seeing this error right now which might relate to what you're saying but I don't know how to fix it.
```
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0x4 (DOWNLOAD(USB/UART0/1))
waiting for download
����������������������������������������������������������������E (1492) esp_littlefs: ./components/esp_littlefs/src/littlefs/lfs.c:1347:error: Corrupted dir pair at {0x0, 0x1}

E (1492) esp_littlefs: mount failed,  (-84)
E (1495) esp_littlefs: Failed to initialize LittleFS
```

If I push the reset button the board, I get this:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x25f4
entry 0x403cc710
```

Then the I configure the Wifi, the device reboots, outputs this:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403820f8
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x25f4
entry 0x403cc710
```

And then I can't get back to the WebUI. Again, nothing else connected to it except for the power supply and USB.

I then followed the same process for the 8266 and get this output:
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v0006b3f0
~ld
```

And seems to repeat, as if in a reboot loop.

csprocket777, 2024-01-13:
I uncommented the `Serial.begin/println` lines to see if I could get more information.

For the 8266, after resetting, reflashing and reconnecting to the Wifi, I get this:
```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

Exception (4):
epc1=0x4000e2ac epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffe00 end: 3fffffd0 offset: 0160
3fffff60:  00000000 00000000 00000000 40210431  
3fffff70:  4022253c 00000000 00001388 40221308  
3fffff80:  00000000 0000002c 3fff226c 00000000  
3fffff90:  feefeffe feefeffe feefeffe 3fff12a8  
3fffffa0:  3fffdad0 00000000 3fff0ee5 402105ce  
3fffffb0:  3fffdad0 00000000 3fff127c 4021e0b8  
3fffffc0:  feefeffe feefeffe 3fffdab0 40100f3d  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

and 
```
mounting FS...
Config loaded
{"on":false,"bri":100,"ct":200}
```

When doing the same on the ESP32-C3, I get:
**Before configuring wifi, first boot after flashing:**
```
��������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������mounting FS...
E (1494) esp_littlefs: ./components/esp_littlefs/src/littlefs/lfs.c:1347:error: Corrupted dir pair at {0x0, 0x1}

E (1494) esp_littlefs: mount failed,  (-84)
E (1498) esp_littlefs: Failed to initialize LittleFS
Failed to mount file system
Create new file with default values
Failed to open config file for writing
Failed to load config
```
**Tried joining it's AP to configure:**
```
E (63247) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (63247) task_wdt:  - IDLE (CPU 0)
E (63247) task_wdt: Tasks currently running:
E (63247) task_wdt: CPU 0: loopTask
E (63247) task_wdt: Aborting.

abort() was called at PC 0x4202a5fd on core 0
Core  0 register dump:
MEPC    : 0x40382520  RA      : 0x403874fc  SP      : 0x3fc93ab0  GP      : 0x3fc8e800  
TP      : 0x3fc7db2c  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x3fc93adc  S1      : 0x3fc93adc  A0      : 0x3fc93ae8  A1      : 0x3fc93aca  
A2      : 0x00000000  A3      : 0x3fc93b15  A4      : 0x00000001  A5      : 0x3fc97000  
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3fc9749c  S3      : 0x7fffffff  
S4      : 0x3fc8fd08  S5      : 0x3fc8fcf8  S6      : 0x3fc8fd00  S7      : 0x3fc8fcf8  
S8      : 0x00000000  S9      : 0x00000001  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
3fc93ab0: 0x00000000 0x00000001 0x3fc93ac8 0x4038c9d2 0x3fc8fd00 0x3fc8fcf8 0x3fc80030 0x3fc90f64
3fc93ad0: 0x3fc93adc 0x3fc90f80 0x3fc93ac8 0x32303234 0x64663561 0x00000000 0x726f6261 0x20292874
3fc93af0: 0x20736177 0x6c6c6163 0x61206465 0x43502074 0x34783020 0x61323032 0x20646635 0x63206e6f
3fc93b10: 0x2065726f 0x00000030 0x3c0b0000 0x2cbb0952 0x600c2000 0x3c0bb000 0x3fc97000 0x4202a600
3fc93b30: 0x00000000 0x00000001 0x00000019 0x0000f70f 0x3fc97000 0x3fc91f80 0x00000000 0x3fc8e1b4
3fc93b50: 0x00000002 0x3fc9b2c8 0x00000000 0x00000001 0x00001881 0x80000007 0x3fc9b2b8 0x403801ee
3fc93b70: 0x00000001 0x3fc93b78 0xffffffff 0x3fc9c900 0x3fc9c900 0x00000001 0x3fc93b8c 0xffffffff

```

**Push the RST button:**
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x25f4
entry 0x403cc710
E (184) esp_core_dump_flash: Core dump data check failed:
Calculated checksum='e58d608f'
I��
mounting FS...
Config loaded
{"on":false,"bri":100,"ct":200}
```

And now I can't get to the WebUI for the ESP32-C3...

csprocket777, 2024-01-13:
And now with the 8266 I'm getting:
```
"/Users/[removed]/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1/python3" -I "/Users/[removed]/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/tools/upload.py" --chip esp8266 --port "/dev/cu.usbserial-10" --baud "921600" "" erase_flash --before default_reset --after hard_reset write_flash 0x0 "/private/var/folders/g1/pkdwgwq57135gknzdvy_h0tr0000gn/T/arduino/sketches/332371455764F4704E8E0D422D0EA2FD/Hue_Gradient_Lightstrip_SK6812.ino.bin"
esptool.py v3.0
Serial port /dev/cu.usbserial-10
Connecting....
WARNING: This chip doesn't appear to be a ESP8266 (chip magic value 0x1b31506f). Probably it is unsupported by this version of esptool.
Chip is ESP8266EX
Features: WiFi
WARNING: Detected crystal freq 19.93MHz is quite different to normalized freq 26MHz. Unsupported crystal in use?
Crystal is 26MHz
MAC: d2:bf:75:94:3a:8b
Uploading stub...
Running stub...

A fatal esptool.py error occurred: Invalid head of packet (0x47)
```

mariusmotea, 2024-01-13:
I think you found the issue.

E (1494) esp_littlefs: mount failed,  (-84)
E (1498) esp_littlefs: Failed to initialize LittleFS

This is caused most likely by wrong specification of the esp board (ex: you have 1MB of SPI flash but you select a board with 4MB of SPI flash).

csprocket777, 2024-01-13:
Except both boards specify 4mb of SPI flash...

csprocket777, 2024-01-14:
So I'm having some success with the 8266 module by upping the `lightsCount` to `6` and `pixelCount` to `300`. 

But I'm getting an exception and a boot loop:
```
ESP Exception Decoder
Sketch: Hue_Gradient_Lightstrip_SK6812 FQBN: esp8266:esp8266:nodemcuv2

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

Exception (4):
epc1=0x4022cbf1 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffe00 end: 3fffffd0 offset: 0160
3fffff60:  00000000 00000000 00000000 402104d1  
3fffff70:  402225fc 00000000 00001388 402213c8  
3fffff80:  00000000 424c0000 3fff22dc 00000000  
3fffff90:  feefeffe feefeffe feefeffe 3fff1318  
3fffffa0:  3fffdad0 00000000 3fff0f55 40210666  
3fffffb0:  3fffdad0 00000000 3fff12ec 4021e178  
3fffffc0:  feefeffe feefeffe 3fffdab0 40100f3d  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception 4: Level1Interrupt: Level-1 interrupt as indicated by set level-1 bits in the INTERRUPT register
PC: 0x4022cbf1
EXCVADDR: 0x00000000

Decoding stack results
0x402104d1: esp8266webserver::ESP8266WebServerTemplate<WiFiServer>::handleClient() at /Users/[removed]/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/ESP8266WebServer/src/ESP8266WebServer-impl.h:282
0x402213c8: std::_Function_handler<void at ), void (*)()>::_M_invoke(std::_Any_data const&) (/Users/[removed]/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:289
0x40210666: loop() at /Users/[removed]/Downloads/Hue_Gradient_Lightstrip_SK6812/Hue_Gradient_Lightstrip_SK6812.ino:1018
0x4021e178: loop_wrapper() at /Users/[removed]/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/cores/esp8266/core_esp8266_main.cpp:258

Paste exception to decode...
```

csprocket777, 2024-01-14:
And...
```
ESP Exception Decoder
Sketch: Hue_Gradient_Lightstrip_SK6812 FQBN: esp8266:esp8266:nodemcuv2

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

Exception (4):
epc1=0x4000c505 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffdd0 end: 3fffffd0 offset: 0160
3fffff30:  00000000 00000000 00000000 3fff118c  
3fffff40:  3fffdad0 00000003 3fff118c 3fff118c  
3fffff50:  00000033 00000044 00000001 402080de  
3fffff60:  00000000 00000000 00000000 402104d1  
3fffff70:  402225fc 00000000 00001388 402213c8  
3fffff80:  00000000 0000002c 3fff22dc 00000000  
3fffff90:  feefeffe feefeffe feefeffe 3fff1318  
3fffffa0:  3fffdad0 00000000 3fff0f55 40210666  
3fffffb0:  3fffdad0 00000000 3fff12ec 4021e178  
3fffffc0:  feefeffe feefeffe 3fffdab0 40100f3d  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception 4: Level1Interrupt: Level-1 interrupt as indicated by set level-1 bits in the INTERRUPT register
PC: 0x4000c505
EXCVADDR: 0x00000000

Decoding stack results
0x402080de: lightEngine() at /Users/[removed]/Downloads/Hue_Gradient_Lightstrip_SK6812/Hue_Gradient_Lightstrip_SK6812.ino:456
0x402104d1: esp8266webserver::ESP8266WebServerTemplate<WiFiServer>::handleClient() at /Users/[removed]/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/ESP8266WebServer/src/ESP8266WebServer-impl.h:282
0x402213c8: std::_Function_handler<void at ), void (*)()>::_M_invoke(std::_Any_data const&) (/Users/[removed]/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/xtensa-lx106-elf/include/c++/10.3.0/bits/std_function.h:289
0x40210666: loop() at /Users/[removed]/Downloads/Hue_Gradient_Lightstrip_SK6812/Hue_Gradient_Lightstrip_SK6812.ino:1018
0x4021e178: loop_wrapper() at /Users/[removed]/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/cores/esp8266/core_esp8266_main.cpp:258

Paste exception to decode...
```

csprocket777, 2024-01-14:
Trying again with the ESP32-C3, same lights / pixelCount as 8266.
```
3fc93cf0: 0x3fc93cf4 0xffffffff 0x3fc93cf4 0x3fc93cf4 0x00000000 0x3fc93d08 0xffffffff 0x3fc93d08
3fc93d10: 0x3fc93d08 0x00000000 0x3fc93d1c 0xffffffff 0x3fc93d1c 0x3fc93d1c 0x00000000 0x3fc93d30
3fc93d30: 0xffffffff 0x3fc93d30 0x3fc93d30 0x00000000 0x3fc93d44 0xffffffff 0x3fc93d44 0x3fc93d44
3fc93d50: 0x00000000 0x3fc93d58 0xffffffff 0x3fc93d58 0x3fc93d58 0x00000001 0x3fc93d6c 0xffffffff
3fc93d70: 0x3fc9bd60 0x3fc9bd60 0x00000000 0x3fc93d80 0xffffffff 0x3fc93d80 0x3fc93d80 0x00000000
3fc93d90: 0x3fc93d94 0xffffffff 0x3fc93d94 0x3fc93d94 0x00000004 0x3fc93da8 0xffffffff 0x3fca0fb8
3fc93db0: 0x3fc9a85c 0x00000000 0x3fc93dbc 0xffffffff 0x3fc93dbc 0x3fc93dbc 0x00000000 0x3fc93dc8
3fc93dd0: 0x00000000 0x00000000 0x00000000 0x3fc93de0 0xffffffff 0x3fc93de0 0x3fc93de0 0x00000000
3fc93df0: 0x3fc93df4 0xffffffff 0x3fc93df4 0x3fc93df4 0x00000001 0x00000001 0x00000000 0x0001ffff
3fc93e10: 0x00000000 0xb33fffff 0x00000000 0x00000000 0x3fc93e1c 0x00000000 0x00000000 0x00000000
3fc93e30: 0x3fc93e34 0xffffffff 0x3fc93e34 0x3fc93e34 0x00000000 0x3fc93e48 0xffffffff 0x3fc93e48
3fc93e50: 0x3fc93e48 0x00000001 0x00000001 0x00000000 0x0001ffff 0x00000000 0xb33fffff 0x00000000
3fc93e70: 0x00000009 0x3fc98ad4 0x3fc98b3c 0x3fc98ba4 0x00000000 0x00000000 0x00000001 0x00000000
3fc93e90: 0x00000000 0x00000000 0x4209fd22 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: 1b36333a2092f343

Rebooting...
```

mariusmotea, 2024-01-19:
If you use our firmware for esp8266 using the web flasher tool you have the same issue? If not the issue is clearly a library used in your Arduino.

csprocket777, 2024-01-20:
I do have the same issue as with the web flashed version. I think it has to do with how loop it takes in the loop with the HTTP server.

csprocket777, 2024-01-20:
I just reflashed the board with the web installer. 

Using the "GRADIENT-LIGHTSTRIP" type. Erasing flash before write. 

It completes. 
I connect it to WiFi and I can access the webUI to configure it. 

It defaults to 60 pixels, I can change it to 250 and it's fine. But my strip has 300. When I configure it to use 300, it gets into a reboot loop. 

[image]

[image]

Using. 5V 20A power supply. Nothing else connected during flash time. 

I've tried using the Arduino ide to flash it in order to tweak different params but everything ends up in the same reboot loop for the same reason. 

I'm lost.

mariusmotea, 2024-01-21:
From what you describe it looks like the leds count variable is one one byte ( 0<->255), but i look now at the code and the type is uint16_t (2 bytes).

https://github.com/diyhue/Lights/blob/master/ESP8266/Hue_Gradient_Lightstrip_WS2812/Hue_Gradient_Lightstrip_WS2812.ino#L44C1-L44C26

One possible issue may be here:

https://github.com/diyhue/Lights/blob/master/ESP8266/Hue_Gradient_Lightstrip_WS2812/Hue_Gradient_Lightstrip_WS2812.ino#L391

instead of for `(uint8_t pixel = 0; pixel < pixelCount; pixel++) {` may be required to set `(uint16_t pixel = 0; pixel < pixelCount; pixel++) {` This loop with every segment led and if you have one segment with more than 300 leds it may cause some issue.

csprocket777, 2024-01-22:
This worked! It's stable now!  Thank you!

The only issue now is, the Philips Hue app doesn't allow me to input more than one color for this gradient strip.
[image]

I'm exploring various updates to see if the situation has changed but figured you'd want to know that this change helped in this case.

mariusmotea, 2024-01-22:
How version of the hue app you are using? Also is Android or iPhone?

csprocket777, 2024-01-27:
Hue App: 5.9.0 (15759)
iPhone

Also, Hyperion only sees 10 hardware LEDs. How do I fix that?

csprocket777, 2024-01-27:
When I configure the entertainment area using the Hue app on the diyHue bridge, even though it sees it as a gradient strip, the light configuration is as if it's simply a horizontal bar. Is this why it's only advertising 10 hardware LEDs?

[image]

csprocket777, 2024-01-27:
Ok. I think I had the wrong .ino file. I ended up using the [Gradient_TV_Strip_WS2812](https://github.com/diyhue/Lights/blob/master/ESP8266/Gradient_TV_Strip_WS2812/Gradient_TV_Strip_WS2812.ino) file *(even though my LED strip is `sk2812`)* and now Hue entertainment and Hyperion see it as the Hue Play Gradient Strip.

csprocket777, 2024-01-28:
...I'm a fool. They aren't interchangeable...

csprocket777, 2024-01-28:
Ok. This seems to work. 

Would I be committing Blasphemy if I simply modified the `server.on("/detect"...)` function of the `Hue_Gradient_Lightstrip_SK6812.ino` file, changing the `root["modelid"] = "LCX004";` line to `root["modelid"] = "LCX002";`?

In my initial tests, it seems to sync and work properly.

mariusmotea, 2024-01-28:
Must not be an issue, if you change the modelid from the web interface Lights page it will rollback to LCX004 on first light scan?

csprocket777, 2024-01-28:
I'm changing what it reports back in the .ino file and it seems to stick. Just can't control the lights after I disable the sync mode in hue.

csprocket777, 2024-01-28:
ok.... I solved the control issue I was experiencing.

I was doing some extra logging in the IDE and noticed that when I toggled `LED Output` off in the Hyperion LED Output instance to turn off the Sync operation, in the `server.on("/state", HTTP_PUT...` handler where it processes the `gradient` key that because the `root["gradients"]["points"].size()` value was `0`, it was setting `lightsCount` also to 0.

This was then persisted which meant the rest of the logic wouldn't iterate over the lights anymore.

## UGLY HACK INCOMING....
In the `state` PUT handler, instead of basing `lightsCount` always on the `gradients.points.size()` value, I'm doing this:
```
      if (root.containsKey("gradient")) {
        if (root["gradient"].containsKey("points")) {
          if (root["gradient"]["points"].size() == 0){
            lightsCount = 7;
          }else{
            lightsCount = root["gradient"]["points"].size();
          }
```

If the size === `0`, I'm just resetting it to `7`, otherwise I'm respecting the `size()` value.

#### Only remaining "boggle"
I really wish I had multi-point color control inside the hue app and I haven't sorted that out yet...

mariusmotea, 2024-01-29:
It is not better to ignore the request when points.size() = 0 ? I'm thinking that you may have a different number of segments and this will always reset to 7. For me the Gradient control is working from both Hue app (v5.9.0) and Hue Essentials on Android. In the Hue app you must have a number on the color picker and long press on it will split into multiple color pickers.

csprocket777, 2024-01-29:
How do I report the proper number of segments? Is this something I need to configure on the light?

csprocket777, 2024-01-29:
I see now how to include multiple colors in the hue app. not intuitive at all... lol.

csprocket777, 2024-02-10:
So I'm struggling a little bit. While this technically works, the experience isn't great. I get all sorts of random flickering and a hard line between sections of the strip. I'm not entirely sure where to go with it.

mariusmotea, 2024-02-11:
Ensure you have a correct division of leds / segments, some errors may occur at the logic operations to find the place of every led in the correct segment. I recommend to start with the default configuration (60 leds / 3 segments) to see if there is any issues, then increase in steps.

csprocket777, 2024-02-12:
Thanks. I'm still looking through this for subsequent lights that I make with the hardware I have but ultimately for this first light I went a different direction/ The Philips Hue arrangement, even if working fully, wasn't going to give me the experience I was looking for.

closed 2024-02-12 by csprocket777