diyhue/Lights #149: Calibrating an ESP8266 Bulb

herobrine30396, 2024-08-05:
I've used a cheap Wipro bulb and converted it to an ESP8266 DiyHue bulb. The problem is colour reproduction. Red is way too dim, and Blue is too bright. I've tried using `b = b * 50 / 100` and `g = g * 80 / 100` as suggested in Slack, but this only works up to an extent. With this change, selecting the colour in the bulb's web interface at full brightness yields the correct colours, but playing with the brightness seems to mess it up. When I tried using Hue Sync for Windows, the colours shown in the app were not what the bulb emits. I've also noticed that the colours were wrong when using Amazon's Alexa, and the Blue channel was too bright even with the code modification. I think there should be a way to calibrate or tune a device for accurate colour reproduction since each light is different.

hendriksen-mark, 2024-08-05:
Hi, in the webui for the light you can select a multiplier.
But if 50% does not work then maybe something is wrong with the light.[image]

herobrine30396, 2024-08-05:
But this only works for WS2811 lights. The light bulb I'm using uses an RGB-CCT circuit.

hendriksen-mark, 2024-08-05:
Oke i will look into this to make them the same webui.

herobrine30396, 2024-08-05:
Okay, thanks!

hendriksen-mark, 2024-08-05:
i did not make the same webui but added rgb_multiplier to the sketch. please try it and give feedback.

herobrine30396, 2024-08-07:
Okay. I uploaded the code, and now the RGB multiplier works. Blue works perfectly at 60, and Red works perfectly at 100, but there's a problem with the Green. Now, I have to set the Green to 10 to get accurate colours, but with the previous code, 70 was enough to do this.

herobrine30396, 2024-08-07:
I've tried changing the gamma correction to 
```
r = r <= 0.04045f ? r / 12.92f : pow((r + 0.055f) / (1.0f + 0.055f), 2.4f);
g = g <= 0.04045f ? g / 12.92f : pow((g + 0.055f) / (1.0f + 0.055f), 2.4f);
b = b <= 0.04045f ? b / 12.92f : pow((b + 0.055f) / (1.0f + 0.055f), 2.4f);
```
and now the RGB multiplier works better. With this formula, Blue works at 90, Red works at 100, and Green works at 30. A problem I noticed with this was that setting the colour picker to white gives a more purple colour than white, but it works with every other colour in the picker.

mariusmotea, 2024-08-07:
Every RGB light is different, the gamma correction we use is generic. Normally with a good balanced RGB light you don't need to change anything. A light manufacturer need to tune this formula after every led type change.