diyhue/Lights #154: Generic_SK6812_Strip: RGB colors multipliers not working

350d, 2024-10-19:
Hello! I've just switched from WS2812 to SK6812 strip and noticed that RGB multipliers can't be used.
All code related to `rgb_multiplier` are missed from sketch.

hendriksen-mark, 2024-10-19:
Hi, thanks for the fix.

closed 2024-10-19 by hendriksen-mark

350d, 2024-10-20:
@hendriksen-mark Do I need to push compiled version for https://install.diyhue.org or it will be available to install automatically?

mariusmotea, 2024-10-20:
The automatic build was turned off since the builded firmware was not working at all. If you can also replace the binary file it will be great.

hendriksen-mark, 2024-10-20:
Please use the new version on github and not your local version, there have been 2 updates to the sketch.

350d, 2024-10-20:
@mariusmotea @hendriksen-mark Thank you for information. Compiled, tested, committed!