diyhue/Lights #159: How get sk6812 more responsive and change colour accurately against Philips tv?

Nicko33127, 2024-11-06:
Hi, 

I managed to build a light strip with esp8266 with 199 LEDs.
If I split this in 5 sections and add this to my Philips tv, the colour change isn't happening fast enough.
Changing this to 3 sections improve slightly but still doesn't match the screen ambilight colours.

I am running diyhue on a mini pc on home assistant.

Am I better of movi g to wled and have one light to have a faster response time as I understand that sections still don't connect well?

Or should I split the strip in 3 physically with three different controllers?

I'll play around over the weekend, but was wondering if I could get some pointers?

Also watching breaking bad the yellow scenes get the strip red instead of yellow, how to adjust this?

Nicko33127, 2024-11-11:
Thank you for the quick answer.

I have now opened port forwarding udc on my router to the IP of my home assistant,but this doesn't seem to improve the speed.

The light changes and it's ok, but nowhere near as fast as the LEDs from the tv. Not sure where to check for the bottleneck.

I am planning to have a 3.5 metre strip currently split into three segments, ideal wanted 5. And a 8.3 meter across 5 bookshelf. With one segment per shelve, but it looks like this will be too much for me to handle ?

hendriksen-mark, 2024-11-11:
You don't need to open a port on your router, opening a port on your router means that you can access it outside your local network.
Please check in the logs if entertainment mode is active, you can also check with hue essentials the cop lights if it works.
You can also upload the config_debug.tar from the webui>diyhue >bridge >bridge control >download debug >full debug, then I can take a look at your config and logs.

Nicko33127, 2024-11-11:
Sorry I got confused with the port, how do I make sure they are open? 

Regarding the entertainment mode, it seem to work when using the hue essential app. i can get the police effect with very quick flashing red and blue.

I can also select each light strips in the hue app and slect where each strips are in the entertainment room option.

I think I might have find the bottleneck. I have an older Philips tv (model 55pus6703/12) that might not support this and only use older http communication and this will be the is the best I can get.