diyhue/Lights #160: Common Anode support for generic RGB strip

MoTastisch, 2025-03-02:
Dear Team, 
I've searched through everything within this project but couldn't find a easy solution to switch the ESP8266/Generic_RGB_Light to a common **Anode** mode. Sure I could fix it with hardware, but it would be easier if a solution from you was provided. Maybe also consider documenting this and the wiring better.

mariusmotea, 2025-03-02:
Hi,

There is an easy way to do this, everywhere you see in the code `analogWrite(pins[color], (int)(light.currentColors[color] * 4.0));` just make it `analogWrite(pins[color], (int)(254 - light.currentColors[color] * 4.0));`

the logic is to replace current value with `254 -` current value because the brightness in hue api is between 1 and 254.

if pwm is 100% duty (254 - 254 brightness => pwm will be set to 0%
if pwm is 0% duty ( 254 - 0 brightness) => pwm will be set to 100%

MoTastisch, 2025-03-02:
Thank you for your fast reply. Maybe you should consider documenting this, as I said, in the documentation. If you wish I could write it as well.

closed 2025-03-15 by MoTastisch