diyhue/Lights #161: Hue_Gradient_Lightstrip_WS2812 does not work for me

sponkyplayer, 2025-04-14:
Hello together,

i flashed the following FW on my Wemos D1 mini. After connecting it to my local WIFI Network it does nothing and the log is empty. 
With the others Hue Firmwares it works properly (Hue_Signe_Gradient, Gradient_TV_Strip_WS2812, etc.). Do some of you have the same problem ?

I would be grateful for an answer.

hendriksen-mark, 2025-04-14:
Hi, did you compile it your self or did you use the flash tool?

sponkyplayer, 2025-04-14:
Hi, i tried both. Compiled by myself with Arduino IDE and with the diyHue ESP Flash Tool. 
But as i said. i am wondering why all the other FWs work. I integrated them into my Hue Sync and they work perfect,
The Generic Strip, where i can configure the zones by myself works, too. But the Gradient Srip does not work after i connect it to my WIFI. after the flashing process.

hendriksen-mark, 2025-04-14:
this weekend i will try to reproduce your issue and look at the code

hendriksen-mark, 2025-04-20:
I find no issue with `Gradient_TV_Strip_WS2812.ino`.(i compiled in vscode)

the library’s i use:
esp8266 board version: 3.1.2 https://github.com/esp8266/Arduino
littlefs version: 1.0.6 https://github.com/lorol/LITTLEFS
neopixelbus version: 2.8.3 https://github.com/Makuna/NeoPixelBus
wifimanager version: 2.0.17 https://github.com/tzapu/WiFiManager
arduionjson version: 6.21.5 https://github.com/bblanchon/ArduinoJson

board config:
board: lolin d1 r2 & mini
CPU freq: 80MHz
VTable: flash
C++ exceptions: disabled
stack protection: disabled
ssl support: all ssl ciphers
mmu: 32KB cache + 32KB IRAM
non-32-bit access: use pgm_read macros for IRAM/PROGMEM
flash size: 4MB(FS:2MB OTA:~1019KB)
iwip variant: v2 lower memory
debug port: disabled
debug level: none
erase flash: all flash contents
upload speed: 921600

the only 2 lines i change are these:
https://github.com/diyhue/Lights/blob/5c7d5816f07d55b7df05a53115db783ecb733042/ESP8266/Gradient_TV_Strip_WS2812/Gradient_TV_Strip_WS2812.ino#L65
https://github.com/diyhue/Lights/blob/5c7d5816f07d55b7df05a53115db783ecb733042/ESP8266/Gradient_TV_Strip_WS2812/Gradient_TV_Strip_WS2812.ino#L579

i changed this to:
65: `NeoPixelBus<NeoGrbFeature, NeoWs2812xMethod>* strip = NULL;`
579: `strip = new NeoPixelBus<NeoGrbFeature, NeoWs2812xMethod>(newCount);`

this is because of the led strip i use.

hendriksen-mark, 2025-04-21:
hi yes im just done with that one, give me a few minutes and i will share my changes for that code

hendriksen-mark, 2025-04-21:
https://github.com/hendriksen-mark/Lights/blob/10ee3d08b916a4366642bbe2b142159fe8975110/ESP8266/Hue_Gradient_Lightstrip_WS2812/Hue_Gradient_Lightstrip_WS2812.ino
can you try this one?

Trenboloni, 2025-04-22:
Hi,

yeah that fix worked for me. Thank you very much !

I just have one more question. I thought three zones could be controlled via this stripe using the entertainment protocol? I only get two zones when integrating them into my Hyperion application via Entertainment. With the Gradient_TV_Strip_WS2812.ino I get 7 zones like with the official Hue Play Grantient Stripe.

hendriksen-mark, 2025-04-22:
Good to hear that this fix the issue, im also looking into that, i only get 1, but now im at work so im not able to look at it.

Trenboloni, 2025-04-22:
Hi,

everything is fine and don't stress. I'm glad you're looking at the subject at all :D

Trenboloni, 2025-07-10:
Hi,

do you know when you can have a look again regarding this topic ?